pioneers/PieCentral #80: Add Protobuf compilers onto beaglebones

brandonxxlee, 2016-11-17:
We currently are committing compiled protobufs into the main repo. These should be instead placed in the gitignore, once we are able to compile the files on the beaglebone.

nikitakit, 2017-02-19:
The devops scripts now handle compiling the protobufs and packaging them into the update tarball. Now that #131 is merged these scripts are run by travis for git tags, uploading the tarballs to github.

There should no longer be a need to put protobuf compilers on the beaglebones. It is also now safe to remove the compiled protos from the repo, except that it should be scheduled to avoid breaking people's development flow.

nikitakit, 2017-03-13:
Closing because travis now takes care of protobuf compilation.

closed 2017-03-13 by nikitakit