pioneers/PieCentral #96: Hosted student API documentation

nikitakit, 2017-01-14:
We previously had student API documentation autogenerated from docstrings using sphinx, and hosted [on readthedocs](http://pie-api.readthedocs.io/). It would be nice to do something similar this year.

I am one of several admins for our readthedocs account/subdomain, and can help if you want to reuse that website.

kevinmasd, 2017-01-30:
We could also package the documentation into Dawn; Dawn already tries to get documentation (though it didn't work last time due to networking).

jiangdaniel, 2017-03-04:
If we put the [API on discourse](https://discourse.pierobotics.org/t/api-spec/281), do we still have to host it on readthedocs too?

mc10, 2017-03-04:
readthedocs (arguably) has a better interface for searching through the docs, so I'd personally say it's useful.

jiangdaniel, 2017-03-04:
Our API is pretty short this year though, so it might not be that big of a deal?  And it's a little bit easier for us to maintain a single copy of the API rather than several copies.

nikitakit, 2017-03-13:
We put an authoritative copy of the API docs up on our website, so I'm now closing this issue.

closed 2017-03-13 by nikitakit