pioneers/PieCentral #104: [Dawn] Dawn UI Update + Notifier

kevinmasd, 2017-01-30:
Might want to find some way to notify user that there is a new version of Dawn or Runtime pushed out.

May expand to actually installing/downloading onto host computer, though this seems very OS dependent.

kev-chien, 2017-02-12:
Ways to do this
- [electron/auto-updater](https://github.com/electron/electron/blob/master/docs/api/auto-updater.md).
- Otherwise, since `bundle.js` and `main.js` are javascript files, consider uploading those files to some cdn and allowing Dawn to download those files when it notices an update is available, since `bundle.js` for instance is loaded in `index.html`.
- The easiest method would probably be to build and upload all new versions of dawn to some database, have Dawn check online for new versions, and then provide a link for downloading another installer/ .app. Less elegant but should work fine.

kevinmasd, 2018-04-29:
End of 2018 Season. If issue re-arises, create a new issue.

closed 2018-04-29 by kevinmasd