pioneers/PieCentral #105: [Dawn] Field Control Variant

kevinmasd, 2017-01-30:
Will need to create a UI for handling game features. Experiment whether it can be done through tabs, windows, or a separate app.

Needs:
- Communication Signals
  - Figure out with networking what we need to send to robots
  - (Basically boils down to coordinating game mode changes)
- Student Locks
  - If student’s Dawn is still open, prevent their version of Dawn from making changes to the robots
- Competition Display
  - Fancy bar mechanism to show timings
  - Show robot-related information to debug/check status

nikitakit, 2017-01-30:
Regarding "student locks", you should talk with field control. I believe the plan is to have field control use a separate wifi network that student machines never have access to.

However, we do ideally want the field control version of Dawn to be able to edit the code running on the robot.

kevinmasd, 2017-02-03:
Wait for #97

kev-chien, 2017-02-09:
My rough schema for Field Control-relevant UI components on student version of Dawn.
See [Dawn View Mock-Up](https://docs.google.com/a/pioneers.berkeley.edu/presentation/d/1nE9djlEVGl7GDGGUdLWshympvxkiGkgxdPfw8p6e6HU/edit?usp=drive_web) for roughly drawn out ideas.

- Make app state variable "GameMode" that tracks whether Dawn app is in game mode (i.e. on the field where it must abide by rules, is connected, displays scores and timer) or in normal editor mode where student can do as pleases.
- Implement new Connections div (i.e. under Peripherals or Gamepads) that displays robot connection and field control connection. Probably will also move run/upload/etc. to here.
- Implement flashy component in "Nav Bar" that switches between Editor and Game Mode info
- Game Mode info component contains timer and scores relevant to game
- Implement Simulate Competition that utilizes Game Mode component.
- Implement switching between Autonomous and Tele-Op (probably in Connections div)
- Change IP button is disabled when in Game Mode, along with probably upload.

kevinmasd, 2017-04-02:
After discussions with others about what Field Control should be doing, it seems likely that the field control version of "Dawn" is going to have to be more barebones than previously though. See [here](https://docs.google.com/a/pioneers.berkeley.edu/presentation/d/1kGP7NuKw6rxlEatfwnkRp1x9FG1sf83ydzgeZ_1l2Pw/edit?usp=sharing) for what I think this variant should look like. Field Station/Regular Dawn is going to have some additional networking modifications then.

kev-chien, 2017-04-02:
So only the Field Control Dawn (not Field Station Dawn) needs the view you drew in the slides? Which would mean a new, separate branch.

dHutchings, 2017-04-02:
Oi.

Question:  Is the second screen in the "Dawn View Mockup" meant to be what
the student sees, or what the staff sees?

Some comments here, assuming it's what the students are supposed to see:

   1. This year, we don't have live scoring.  (We will in future games).
   That means that we won't have any system that will be able to report the
   driver station score in real time, so it may not be useful to display it.
   At least this year.

   2. I was under the impression that the field control dawn would display
   a team's code (in greyed out fashion).  It'd allow teams to know what code
   is actually on their robot without being able to change it.

   Nikita was even suggesting that it may be a good idea to give teams the
   ability to edit their teleop code during the match.

   This is because many teams will edit their code in the queue, and
   inevitably, they'll mess it up.  There will be several instances where
   team's robots won't drive because they edited their code, and it has some
   stupid syntax error.  This change would allow teams to realize this
   problem, and then fix it (on the fly).

   This change might also be valuable, but should be weighed against the
   cost of having teams erroneously press the play button when they're not
   supposed to or something.

   3. I would really appreciate it if you could talk the intended
   functionality of the student-facing driver station through with nikita, and
   possibly others.

Thank you!

-Doug

kevinmasd, 2017-07-30:
#329 merged in, any further discussion can be on individual components.

closed 2017-07-30 by kevinmasd