pioneers/PieCentral #116: Add LICENSE

mc10, 2017-02-09:
Since this project is open source, adding a LICENSE would be beneficial. GPLv3 and MIT are fairly common.

nikitakit, 2017-02-16:
License file is now in the repo (this has been on my TODO for months now, it's just that I've never gotten around to it). Apache 2.0 has been the standard code license for PiE for years now, as you can see from old repos like tenshi or forseti2.

Keeping this open because we should consider adding a license blurb ([like this](https://github.com/pioneers/tenshi/blob/master/build.sh#L3)) at the top of our source files, such that the license information isn't lost as individual files get copied around.

brandonxxlee, 2017-06-07:
@nikitakit Do you know if we should use the header from the one on the LICENSE or something like the one in tenshi. They differ by a little, so not sure which one is correct.

kevinmasd, 2017-07-10:
Is it necessary to put the blurb everywhere? I feel that this would only be needed on code that might be used elsewhere or if parts of the code are licensed differently.