pioneers/PieCentral #120: Dawn: useful features for menu, editor, files, configurations (saved state)

kev-chien, 2017-02-09:
Some nice editor features that shouldn't be too hard to implement
- Have automatic re-opening of last file, e.g. having it saved in persistent local memory
- Have configuration (persistent setting) if students don't necessarily want to see their last file they were working on when they last closed dawn for some reason
- Have menu bar show recently opened files (and easy clicking)
- Have shortcuts for editor features (may open another issue for this)
- Have configuration (persistent setting) if for some reason students don't want to be prompted when they close editor without saving that they didn't save

kevinmasd, 2017-03-16:
I don't know if any of these things are necessary. The saving prompt seems too important to ignore, and automatic re-opening of last file seems odd to me.

kev-chien, 2017-03-16:
True haha, I think these features may not be something students necessarily want. It may be helpful to survey them midway of what features they think they would want and learn about how they use Dawn.
Otherwise, I'll close the issue for now.

closed 2017-03-16 by kev-chien