pioneers/PieCentral #126: [HIBIKE] Filter serial ports by vendor/product ID

nikitakit, 2017-02-15:
Currently the way hibike finds ports to use is the following:

```
ports = glob.glob("/dev/ttyACM*") + glob.glob("/dev/ttyUSB*")
```

This has the side effect of possibly catching non-hibike devices, if they're currently plugged in to the computer. @dHutchings brought up this issue.

I think we can use the [serial.tools.list_ports module](http://pyserial.readthedocs.io/en/latest/tools.html) to find devices while matching for vendor/product ids of the hardware we actually use. This should cut down on the possibility of interfering with random devices.

cc: @karthik-shanmugam

karthik-shanmugam, 2017-02-16:
What should I filter devices by? Is a hard-coded white-list of (vendor_id, product_id) tuples sufficient?

nikitakit, 2017-02-19:
I think the (vid, pid)'s of our Arduinos should be consistent. Ideally production orders would all be coming in from known-good suppliers, and then we can update the whitelist to the most specific setting that matches that supplier.

Other than that, we should keep (vid, pid) of known Arduino development boards in the code as well (i.e. the bigger form-factor boards that we sometimes use for testing)

baby-bell, 2017-10-28:
I suspect that this is not relevant anymore, thanks to changes in the way we detect devices.

closed 2017-10-28 by baby-bell