pioneers/PieCentral #160: [DAWN] Fix performance by consolidating actions

nikitakit, 2017-02-26:
Currently, `Ansible.js` dispatches a number of actions:

```js
RendererBridge.reduxDispatch(ansibleConnect());
RendererBridge.reduxDispatch(updateStatus());
try {
  const data = RuntimeData.decode(msg);
  RendererBridge.reduxDispatch(updateRobotState(data.robot_state));
  for (const sensor of data.sensor_data) {
    RendererBridge.reduxDispatch(updatePeripheral(sensor));
  }
  RendererBridge.reduxDispatch(updateAllPeripherals(data.sensor_data));

} catch (e) {
  console.log(`Error decoding: ${e}`);
}
```

This should be reduced to just two actions: one to update the Info state, and one to update the Peripheral state.

In particular, the `for` loop must be moved into the reducer.

vdonato, 2017-02-26:
+1 pls fix

kevinmasd, 2017-02-26:
Pushed to branch. Want me to PR it?

nikitakit, 2017-02-26:
@kevinmasd Yes, please send a PR

nikitakit, 2017-03-13:
This was fixed a while ago.

closed 2017-03-13 by nikitakit