pioneers/PieCentral #218: [Dawn] Error if no internet connection

nikitakit, 2017-03-08:
I just started Dawn while not connected to wifi, and got an error:

```
Uncaught Exception:
Error: send ENETUNREACH 192.168.0.200:1236
    at Object.exports._errnoException (util.js:1022:11)
    at exports._exceptionWithHostPort (util.js:1045:20)
    at SendWrap.afterSend [as oncomplete] (dgram.js:404:11)
```

I'm running the [latest pre-release Dawn](https://github.com/pioneers/PieCentral/releases/tag/test%2Fdawn%2F2017-03-07).

mc10, 2017-03-08:
This should be fixed by #199 after it gets merged in (after some local testing), but definitely check to see if that's the case.

kevinmasd, 2017-03-16:
Can someone verify if this has been addressed?

kevinmasd, 2017-04-02:
Noticed on one of the chromebooks today, but Dawn worked fine anyway. Might be a Crouton issue?

kevinmasd, 2017-04-10:
Haven't seen this recently, so I'll close until I get more details.

closed 2017-04-10 by kevinmasd