pioneers/PieCentral #219: [HIBIKE] Read queue overflow stalls device writes

nikitakit, 2017-03-08:
We've noticed that time taken to write to a hibike serial port can spike as high as 3 seconds. Close examination has shown that this is because the *read* queue for that port fills up, which prevents new operations for being executed against the device.

This means that `hibike_message.py::blocking_read` cannot effectively keep up with bytes coming in. A hacky fix is 
```diff
diff --git a/hibike/hibike_message.py b/hibike/hibike_message.py
index 7825bdb..f5e25e4 100755
--- a/hibike/hibike_message.py
+++ b/hibike/hibike_message.py
@@ -332,6 +332,10 @@ def parse_bytes(bytes):
 def blocking_read(serial_conn):
   buffer = bytearray()
   while not parse_bytes(buffer):
+    if serial_conn.in_waiting > 500:
+        buffer = bytearray()
+        serial_conn.reset_input_buffer()
+        print('reset input buffer!!!!!!!!')
     curr = serial_conn.read()
     if struct.unpack('<B', curr)[0] == 0:
       buffer = bytearray(curr)
```

However, this is just a hack. There needs to be a real fix implemented.

karthik-shanmugam, 2017-03-09:
Reading definitely needs to be fixed, but I'm curious; What part of the write procedure is being blocked by the read queue being full? I thought Serial communications were supposed to be duplex.

nikitakit, 2017-03-13:
This was fixed in #242

closed 2017-03-13 by nikitakit