pioneers/PieCentral #299: Restart Runtime Button

kevinmasd, 2017-04-02:
Add last season's ability to restart runtime to the new version of Dawn. May need code refactoring, since ssh2 items are probably getting pretty verbose now.

dHutchings, 2017-04-05:
Question:

Is this a good idea?  Considering that every other time you try to upload code, dawn shows an error message of "Runtime Unresponsive", is it a good idea to give teams a button to restart runtime?

We should aim to hide as much complexity as possible, and giving teams the ability to restart run time at whim seems like a recipe for bad things...

kevinmasd, 2017-04-05:
We provided this ability last year, which apparently did fix things anecdotally. Anyway, one of the most common way to fix things is to reboot everything anyway. Currently, that means power cycling the robot, which takes longer to do so than just sending the command to restart runtime.

closed 2017-04-09 by kevinmasd