pioneers/PieCentral #348: [HIBIKE] Update Makefile to be smarter about builds

baby-bell, 2017-06-24:
Currently, whenever a rebuild is necessary, the Makefile nukes every object file and rebuilds them all from scratch. This is probably unnecessary, and it slows things down a lot.

dHutchings, 2017-11-03:
Indeed.  There's some interesting information behind this that you may want to know about.

1) The flash script nukes the object files by rm -r(f?)ign them.  This is deliberate.  I couldn't get "make clean" to work last year. 

2) We cannot use the same object files each time (without edits), because doing so would lead to flashing successive devices as the same UID.  The most immediate way we got around this was by forcing a recompilation from scratch.

3) It was at least proposed last year that we could figure out what lines of the hex file contain the UID, and put in new UIDs there.  That would at least speed up the mass-flashing.

4) I am not good enough with makefiles to know if there are other options.  The fact that "Make clean" doesn't work is... very suspicious.  If it did work, then we would be able to leverage make voodo to recompile only small portions of the code.

tingtingchiang, 2018-01-25:
@dHutchings do you remember what make clean used to be? Right now it looks like it rm -rf OBJDIR, which is all of the compiled files anyways.

baby-bell, 2018-03-18:
I did a little bit of investigation. Forgive me if you already knew this.
- The UID is defined as a C macro on lines 69 and 70 as `UID_RANDOM`, and every library has access to it 
- Only Hibike actually uses the macro
- The compiled binary is in `bin/hibike/micro` and the compiled library in `bin/hibike/micro/userlibs/hibike`

This suggests to me that we could potentially just delete the compiled binary (the `.hex`, `.elf`, and `.sizeok` files) along with the hibike library objects, and it would still work. At the very least, Make doesn't rebuild everything else when you do this.

PR #570 (2018-03-22): [HIBIKE] Don't recompile all source files on successive flashes
Closes #348
closed 2018-03-22 by baby-bell