pioneers/PieCentral #354: [HIBIKE] Hotplugging

brandonxxlee, 2017-07-06:
Currently hibike/runtime does not support hotplugging.

baby-bell, 2017-07-14:
This also requires some support from Dawn.

tingtingchiang, 2017-07-14:
A place to start would be to look into dmesg, and grepping information from there since Linux itself detects if something gets plugged in.

baby-bell, 2017-07-19:
Yeah, looks like dmesg does show when a device is detected. I am a little bit worried about chaining the implementation to linux only, though. Might just try rescanning ports periodically.

tingtingchiang, 2017-07-20:
That's true, it would also be a lot easier to figure out. Though, are we planning on moving off of linux? I can't imagine running Mac or Windows on a BBB

bodiddles, 2017-07-20:
Considering a lot of the stack relies on us having Linux on the BBB, I think it's fine to "chain" ourselves to linux. If we were to switch the OS, there would be a lot of other refactoring that would have to be done.

dHutchings, 2017-07-20:
I'm not so sure about a dmesg-based solution.  For the robot, we'll
probably be sticking with linux for the forseeable future, no matter if
we're on the BBB or some other SBC.

However, specifically tying the hibike stack to dmesg (and hence, linux)
will limit developer's options.  Right now, hibike_process.py can
forseeably run on Mac (thanks to Ivan), and run on windows with very little
changes.  (Namely, just changing "dev/ttyACM*" to "COM*").  A dmesg based
solution would preclude that option, unless we figured out the mac and
windows equivalent of dmesg.

tingtingchiang, 2017-07-20:
Makes sense, it also slipped my mind that everything is done in python/C++ (been staring at bash scripts way to long oops). dmesg works on mac, and I don't think the current code works on windows anyways. Brose's solution sounds good though, do we have an idea on how often we want to check? And how much processing power that'd take?

dHutchings, 2017-07-20:
Well, we get to choose the rate.  It is true that we're busy-waiting here,
but if we do so every ~250 or ~500 ms, it shouldn't be a problem.

I think? that hibike would actually work on windows, if we added "COM*" to
the code.  But that's either here nor there.

nikitakit, 2017-07-21:
You may also want to consider using the linux udev system to notify you whenever devices are plugged in (or unplugged). This has the benefit of more quickly detecting new devices: I suspect that a ~250ms polling rate will introduce a quite noticeable delay, while a faster rate may use up too much CPU resources. An event-based system like udev has the advantage that code only runs when there is actually an event to respond to.

I think it's fine for the hotplugging features to be linux-specific (it may be possible to still support other systems, just disabling the udev-based hotplugging). For example runtime only runs on Linux (and not even Mac, for some odd reason).

Links:
* http://pyudev.readthedocs.io/en/latest/api/pyudev.html#pyudev.Monitor
* https://askubuntu.com/questions/508236/how-can-i-run-code-whenever-a-usb-device-is-unplugged-without-requiring-root

baby-bell, 2017-07-21:
I think that's an interesting idea. I'll have to see how bad the latency on polling can get; as it is now, detecting a device can take a few seconds anyways, so 250ms doesn't seem that bad.

nikitakit, 2017-07-22:
If detecting a device really does take a few seconds, then polling shouldn't be a problem. Have you verified that this is actually the case? I was always under the impression that the kernel can detect new devices quite fast, and it was the higher-level code (such as hibike) that introduced the seconds-long delays.

baby-bell, 2017-07-22:
I should clarify; detecting a new __USB__ device takes very little time. Detecting whether that new device is a smart sensor takes a few seconds at most.

dHutchings, 2017-07-24:
To clarify:  "Detecting whether that new device is a smart sensor takes a
few seconds at most."  --> Do you mean using your code that detects whether
a COM port contains a smart sensor via sending messages and seeing if they
give correct responses?

It would be very helpful to decide on a goal here - e.g. "It takes < N
seconds for a new device to be detected".

I suggest that we set 1 second as a goal.

baby-bell, 2017-07-24:
Yes, it is already based on a timeout, for reads and writes. See `IDENTIFY_TIMEOUT` in `hibike_process`.

On another note, there is some work being done on a polling-based system on `hibike/hotplug-poll`.

baby-bell, 2017-08-03:
Has been addressed in #369

closed 2017-08-03 by baby-bell