pioneers/PieCentral #374: [HIBIKE] Yogi Bear Error State Handling

brandonxxlee, 2017-07-30:
Change the Enable pin on Motor controller to go to an ADC pin. Ask Electrical for what pin it goes to. 

The Enable pin is an analog value that can be read based on what state the drier is in. Usually, it is around 5V, but when the motor controller falls into an error state, it will fall to roughly 4.5 volts. This can be read on an ADC pin, and the error state can be either handled in some way, i.e. displaying on Dawn if a motor controller status.

tingtingchiang, 2017-08-05:
Switch pin to read as analog and giving it an error state.

baby-bell, 2018-02-03:
This requires some board changes as well, so probably will not happen. Postponed until next year.

closed 2018-02-03 by baby-bell