pioneers/PieCentral #386: [HIBIKE] Move to protobufs/other RPC library

baby-bell, 2017-08-08:
Currently, we use an ad-hoc system based on struct packing and unpacking. It might be nice to use a purpose-built library for this purpose, as with the use of protobufs by Ansible.

nikitakit, 2017-08-09:
One thing to think about is that PiE often has returning teams who try to plug in smartsensors from previous years into their robots. PiE staff also probably have some old devices left over, too. If there aren't any major electrical changes that would make old devices unusable, I would strongly recommend prioritizing backward compatibility over protocol changes.

This past year there was a major rewrite of hibike that broke protocol compatibility and also created a bunch of merge conflicts (instead of being implemented in a continuous-integration fashion). The nature of these changes caused cascading development difficulties for other projects, such as runtime and certain firmware components. I think PiE should consider how to do a better job for future development. It might even be worth deciding, as on organization, what sort of compatibility level is desired and filing bugs anytime master breaks it.

baby-bell, 2017-08-11:
Yes, I think that protocol compatibility is the most important. I thought it might be possible to describe the protocol we have now using something like protobufs, so that we cut down on the amount of homegrown encoding/decoding code.

dHutchings, 2017-11-03:
Well at the very least we can cook up large if tables, that parse messages differently based upon the specific protocol tat

karthik-shanmugam, 2017-11-03:
Is there any issue with the current protocol? Protobuf will not work on Arduino (I think it uses the heap for parsing) and even flatbuffers will likely have too much overhead. I'm not sure what assumptions about data integrity either one uses / how good usb-serial data integrity is. In any case both of these protocols will require packet framing since usb-serial is a byte stream, and we would still need a protocol for that. Perhaps there is a serialization library intended for things like Arduino?

The protocol as it was at the end of last year unfortunately requires complicated encode/decode logic. I don't know if any fully custom serialization protocol can avoid this, especially considering that our protocol also has checksums and cobs encoding for packet framing and data integrity (which might not be real issues if USB does all of this for us btw).

I don't think breaking backwards compatibility is a huge issue in terms of obsoleting last year's hardware since a firmware update can fix this, although I might be underestimating logistical/operational challenges here. The slowdowns in full stack development/integration from such a breaking change were definitely a major issue though.

baby-bell, 2018-02-10:
The existing protocol works, and this is a low enough priority that I'm closing this for now. If it becomes an issue in the future, we can reopen.

closed 2018-02-10 by baby-bell