pioneers/PieCentral #395: [HIBIKE] Add `timestamp` debug command to hibike/runtime

baby-bell, 2017-08-26:
Currently, there is nothing in the way of time information for Hibike and Runtime commands. This greatly increases the pain involved in measuring latency. It would be nice if there was a command to print timestamps at every level.

baby-bell, 2017-09-21:
# Timestamp Specification

tingtingchiang, 2017-09-23:
3. Hibike will receive this command from StateManager and process it in an if/elif/else block that can be found by ctl+f "enumerate_all" (another command sent from StateManager). 
We should figure out how to return the information, either through another message packet/pipe or as another "device". I'm personally for device

baby-bell, 2017-09-23:
The chain goes in reverse, we send back to state manager through a pipe. We should have two types of commands, to indicate whether it's coming from StateManager or Dawn.

baby-bell, 2017-10-28:
Implemented in #430

closed 2017-10-28 by baby-bell