pioneers/PieCentral #417: API Autocomplete

kevinmasd, 2017-10-07:
The [React-ACE](https://github.com/securingsincity/react-ace), the library currently used as the editor, does support basic autocompletion features through its underlying ACE library, seen [here](https://github.com/ajaxorg/ace/wiki/How-to-enable-Autocomplete-in-the-Ace-editor). Adding our own API into the autocomplete however doesn't seem too easy.

Alternative editors, like [Draft.js](https://draftjs.org/) or [React-Quill](https://github.com/zenoamaro/react-quill), also don't seem to make autocompletion easy.

ArjunSridhar1, 2017-10-18:
Currently held up by lack of documentation for snippets and a lack of JSON for PiE API

ArjunSridhar1, 2017-10-25:
Decided to hardcode in the API because it is relatively stable.

closed 2017-10-28 by ArjunSridhar1