pioneers/PieCentral #446: [HIBIKE] Use `asyncio`

baby-bell, 2017-10-30:
Realistically, the way HIbike does IO right now would be better suited for async io. This is super far off in the future probably.

karthik-shanmugam, 2017-10-30:
Is it possible to convert parts of hibike to asyncio while keeping the rest synchronous and threaded?

baby-bell, 2017-11-04:
It's difficult, because HIbike is built on the assumption that threads are being used. There are a bunch of
explicit event loops that would need to be converted to coroutines.

tingtingchiang, 2018-02-03:
let's use hibike/asyncio as master branch for any changes.

tingtingchiang, 2018-02-03:
- [ ] Transport write methods (through coroutine)
- [ ] Read (can be overwritten)
- [ ] Asyncio and pipe interface (read and write from hibike to sm)
- [ ] Integrate hotplugging into async
- [ ] How to keep coroutines running in the loop as a loop (while true + await)
- [ ] How to stop looping coroutines from running again
- [ ] Exceptions behavior