pioneers/PieCentral #451: [HIBIKE] Remove unused `deadband` parameter from YogiBear

baby-bell, 2017-11-04:
We don't actually allow students to read or write this value, so might as well remove it.

dHutchings, 2017-11-07:
Or we might as well provide better documentation as to what it does, and then allow students to write to it.

This follows up on a larger theme: there's lots of knobs on the yogibear, and the documentation as to what they do on the student end is atrocious.  For example, no teams used position PID last year, even though it would have been extremely helpful for them.

The deadband parameter will help remove that really annoying "buzz" when the joysticks are at zero.  This occurs (or rather used to - I assume this is still the case) because xbox controllers only report new values when changes occur, so it's possible that the control system will command the yogibear at 1% PWM for extended periods of time.

dHutchings, 2017-11-07:
maybe I should rephrase: it's not atrocious (esp compared to years past), but it's definately unusable.

PID and Current Limits are hard things to explain in a block paragraph form. (https://pioneers.berkeley.edu/software/robot_api.html#robot-set-value-device-name-param-value-yogibear)