pioneers/PieCentral #654: [DAWN] Version Compatibility

lawrencechen14, 2018-11-10:
- Currently, Dawn only accepts Node versions 4 through 9, but now since Node version 10 is the standard, we need to make Dawn compatible with these newer versions.
- Make changes to allow Dawn to accept Node versions >=10 and ensure that running Dawn on Node version 10 does not cause anything to break.

lawrencechen14, 2018-11-10:
The dependency that is breaking it is the upath package in node_modules, which expects a node version <=9. However, it seems like this local upath package is out of date since the one in github has no upper bound. Figure out how to update all of the node_modules to all be up-to-date.

Aakup, 2018-11-14:
yarn config set ignore-engines true
yarn upgrade left-pad@^1.0.0

closed 2019-03-16 by lawrencechen14